wiringX/wiringX #1: How to distinguish a BananaPi from a Cubietruck?

gkoala, 2015-01-07:
Both boards feature an A20 SOC and the current identification code in WiringX/bananapi.c only checks for that. But is there even a reasonable way of distinguishing those boards?

Unfortunately, it is necessary to separate them because of incompatible GPIO numbering, ...

CurlyMoo, 2015-01-07:
What is the output of cpuinfo in the Cubieboard/

CurlyMoo, 2015-01-08:
I've got another idea. Can you post the mac address of the LAN interface of the Cubie?

linux4kix, 2015-01-08:
Sorry to be late to the party.  We should be able to use
/sys/devices/soc0/machine to distinguish boards using the same SOC.

CurlyMoo, 2015-01-08:
Doesn't seem present on the Raspberry Pi and Radxa Rock (Pro).

linux4kix, 2015-01-08:
Do they not use device-tree yet?

CurlyMoo, 2015-01-08:
No

gkoala, 2015-01-08:
Will try to answer those questions tonight as soon as I get to my cubie.

MAC address might be an idea, if there is a common board specific part for all cubies which differs from the bananas. Difficult to determine without documentation and access to a sufficient number of boards, though.

gkoala, 2015-01-08:
cubie:~ > cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 4 (v7l)
processor       : 0
BogoMIPS        : 956.47

processor       : 1
BogoMIPS        : 832.55

Features        : swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 4

Hardware        : sun7i
Revision        : 0000
Serial          : 0000000000000000
Chipid          : 16516649-80485471-52564848-08c0b591

==================================================================
root@cubie:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 02:49:08:c0:b5:91
          inet6 addr: fe80::49:8ff:fec0:b591/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:580 (580.0 B)
          Interrupt:117 Base address:0xc000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 98:3b:16:de:c7:c6
          inet addr:192.168.178.28  Bcast:192.168.178.255  Mask:255.255.255.0
          inet6 addr: fe80::9a3b:16ff:fede:c7c6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:816 errors:0 dropped:0 overruns:0 frame:0
          TX packets:344 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:70376 (68.7 KiB)  TX bytes:37283 (36.4 KiB)

============================================================
/sys/devices/soc0 doesn't exist

CurlyMoo, 2015-01-08:
Can you please use markup codes for better readability?

smutbert, 2015-01-13:
Afaik the Banana Pi doesn't have a S/PDIF output. So i would suspect the device sunxisndspdif should only appear on the Cubietruck.

```shell
$ cat /proc/asound/cards
 0 [sunxicodec     ]: sunxi-CODEC - sunxi-CODEC
                      sunxi-CODEC  Audio Codec
 1 [sunxisndspdif  ]: sunxi-sndspdif - sunxi-sndspdif
                      sunxi-sndspdif
 2 [sunxisndi2s    ]: sunxi-sndi2s - sunxi-sndi2s
                      sunxi-sndi2s
```

i don't know if this is a good idea though...

vitalogy, 2015-08-05:
Are there any news to get this work on a CubieTruck?
Running here a CubieTruck with Arch Linux on kernel-4.2-rc5 as a little server.
So i would also like to run pilight on this device, and pilight uses wiringX, but at this time there is no support for CubieTruck (CubieBoard3) in wiringX.

Could the following help?

    # cat /sys/firmware/devicetree/base/model
    Cubietech Cubietruck

Kernel-4.2-rc5 uses a devicetree.

smutbert, 2015-08-05:
most users will still run kernel 3.4.x I think. At least I do since the s/pdif-jack was my reason for choosing the cubietruck and sound was not (yet?) supported with more recent kernel versions last time I checked.

Of course it would be great if wiringX supported the cubietruck. gkoala, who started this issue, already did a good job porting wiringPi/wiringCB to the cubietruck. He posted his work in the german cubieboard-forum:
http://www.forum-cubieboard.de/Thread-wiringcb-wiringpi-und-cubietruck
in this post you find the link to the last version he posted
http://www.forum-cubieboard.de/Thread-wiringcb-wiringpi-und-cubietruck?pid=4399#pid4399

I had a few problems instaling it (i don't have sudo installed and had to do it manually) and in some situations you have to set up a pull-up to get a pull-down and vice versa, but apart from that everything worked fine.
I don't know pilight though and have no idea if it will work too.

CurlyMoo, 2015-08-05:
People should just port directly into wiringX so we have one single library instead of all those individual wiringPi ports.

mindoLife, 2015-08-05:
*You are right*
*I'm using wiringPi and waiting for Pi4J (or other project) to support
wiringX to move. *
*i hope it will be soon*

vitalogy, 2015-08-12:
I think with linux-3.4.x you can not clearly determine which board the user is running.
The only thing you can do is a process of elimination, like it is done [here]
(https://github.com/igorpecovnik/lib/blob/next/scripts/armhwinfo)
With this old kernel you get only a generic description which SoC (sun4i, sun7i, ... ) is used.

For the cubietruck you could check for the totalmem of 2GB. But i don't know if there are other sun7i boards around with a ram of 2GB.

The better solution is to use the description from the devicetree.
You get an unique string by checking

    cat /sys/firmware/devicetree/base/model

or by

    dmesg |grep "Machine model"

I have also take look in bananapi.c, why it checks for sun4i and sun7i?
Afaik BananaPi and BananaPro (if it cheks for that at all) are using A20 SoC (sun7i), only the popular Cubieboard has an A10 SoC (sun4i)!

| Board | SoC | RAM | Pins |
| --- | --- | --- | --- |
| Cubieboard  | A10 sun4i  | 1GB  |  96 external pins |
| Cubieboard2 | A20 sun7i  | 1GB  |  96 external pins |
| Cubietruck |  A20 sun7i |  2GB  |  54 external pins |
| Banana Pi  |  A20 sun7i  | 1GB  |  26 external pins |
| Banana Pro |  A20 sun7i |  1GB  |  40 external pins |

> He posted his work in the german cubieboard-forum:
http://www.forum-cubieboard.de/Thread-wiringcb-wiringpi-und-cubietruck
in this post you find the link to the last version he posted
http://www.forum-cubieboard.de/Thread-wiringcb-wiringpi-und-cubietruck?pid=4399#pid4399

I have seen this, but his port will propably not work with pilight.

CurlyMoo, 2016-05-14:
New rewrite branch has the auto detection removed. Users much tell wiringX what platform they use.

closed 2016-05-14 by CurlyMoo