wiringX/wiringX #2: GPIO numbering in sysfs is user definable on Cubieboards

gkoala, 2015-01-09:
The GPIO numbers on the Cubie are user definable. I.e. if you find a GPIO in /sys/class/gpio/gpio# at all, and under which number you find a particular GPIO there is completely up to the user. There is no default.

As opposed to the other boards, the Cubie exposes this setup to the user via the FEX mechanism.

Since the only apparent way to set up interrupts is through sysfs ("poll" requires a file handle), this is a problem. Any predefined assumption made by wiringX may conflict with the user's setup. Ultimately, it may be necessary to let users define the mapping between wiringX numbers and GPIO numbers.

CurlyMoo, 2015-01-09:
Can you tell me more about that FEX mechanism?

CurlyMoo, 2015-01-09:
Then that's their problem. We can't do much about it. We will create our own mapping and it's up to the users to opt for support @cubieboard.

gkoala, 2015-01-09:
Yep, my thoughts exactly. It's mostly a matter of documentation.

CurlyMoo, 2015-01-09:
Think of something reasonable and just adapt the index.html here with the new information:
https://github.com/pilight/wiringX/blob/gh-pages/index.html

CurlyMoo, 2016-05-14:
You can check the new rewrite branch. That should help you port the Cubieboard much more easily.

closed 2019-11-26 by CurlyMoo