wiringX/wiringX #3: Hummingboard Keypad

johnsonjw1, 2015-01-10:
Has anyone used wiringX on the Hummingboard to configure a 4x3 keypad without using the breadboard?

CurlyMoo, 2015-01-10:
What did you try and what issues did you have while doing it?

johnsonjw1, 2015-01-11:
I've read the basics of wiringPi, connected the keypad to the GPIO pins according to wiringX for my COL(1,4,5,6) and ROW (0,2,3). Downloaded wiringX to Debian Jessie and tried a couple very basic but failed attempts of writing my first few lines of c.  All ID10T errors, I am sure.

Any suggestions or examples that I could take a look at that would help me along? There are plenty of examples and tutorials for the RPi.GPIO and wiringPi with Python, but not much for wiringX in C.

I'm a beginner but willing to learn.

CurlyMoo, 2015-01-11:
wiringX uses the same logic as wiringPi. Just replace the wiringPi names with wiringX (e.g. `wiringPiSetup -> wiringXSetup`). Also check the pin numbering on `http://wiringx.org`

HonesDefwig, 2015-01-27:
Love the new addition of the Python module.

Do you have any plan to add Wiringpi's "pullUpDnControl" to WiringX? That is essential for all the keypad tutorials I have come across.

CurlyMoo, 2015-01-27:
Don't know what it does :)

rabeeh, 2015-01-27:
To control pull ups and pull downs of the i.MX6 pads for example.
i.MX6 can be configured that each of it's pads can be 22k/47k/100k ohm pulled up, or 100kohm pulled down.
It can be configured with hysteresis on/off, open drain, drive strength and other goodies :)

HonesDefwig, 2015-01-27:
Right, it's sets the internal pull-up or pull-down resistor mode on the given pin.

http://wiringpi.com/reference/core-functions/

CurlyMoo, 2015-01-27:
So @rabeeh, nice one for you to implement :+1: Then i will check Raspberry.

closed 2016-05-14 by CurlyMoo