wiringX/wiringX #4: Error when installing python gpio

stefaanbolle, 2015-01-15:
When trying to install the python gpio libraries I get following error:

/opt# python setup.py install
running install
running build
running build_py
running build_ext
building 'wiringX.gpio' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I../ -I/usr/include/python2.7 -c wiringX/wiringx.c -o build/temp.linux-armv7l-2.7/wiringX/wiringx.o -Wformat=0
wiringX/wiringx.c:34:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1

Searching for Python.h, the file is indeed not present in the wiringX folder structure

CurlyMoo, 2015-01-15:
Install python-dev first: `sudo apt-get install python-dev`

closed 2015-01-15 by CurlyMoo

stefaanbolle, 2015-01-16:
Thanks! Working  now!!