wiringX/wiringX #14: How do you use PWM?

skpacman, 2015-02-25:
I have a LCD screen where the backlight is actually 3 separate LED's (RGB) that you can either PWM or adjust the voltage to change the color combination.

The problem is, the Anode is the common side, not the cathode. So a PWM_OUTPUT wouldn't do much good, I need to PWM an INPUT. Is that possible with wiringX? Otherwise, I'll just stick with using potentiometer's to control the screen's color...

CurlyMoo, 2015-02-25:
Can you post pseudo code of what you want to achieve?

skpacman, 2015-02-25:
Not sure what you mean? I don't even know how to use the PWM_OUTPUT yet, I was just looking through your code to see what's possible.

CurlyMoo, 2015-02-25:
There is no `native` PWM implementation. You should code that yourself.

skpacman, 2015-02-25:
You do have PWM_OUTPUT set up in your [wiringX.h](https://github.com/wiringX/wiringX/search?utf8=%E2%9C%93&q=PWM), or is that just a placeholder?

I'd like to be able to set the Hz at which it modulates each output/input if possible like:
```
gpio.pinMode(gpio.PIN8, gpio.PWM_INPUT) #set Pin 8 mode to PWM input
gpio.pwmSet(gpio.PIN8, 60) #set Pin 8 PWM to 60Hz
```

CurlyMoo, 2015-02-26:
That's a placeholder.

closed 2015-05-05 by skpacman