wiringX/wiringX #16: Document requirements for adding platforms

crazystick, 2015-03-03:
It would be nice to have a step-by-step guide for adding new platforms, to avoid breaking things:

* add implementation
  * minimum set of functions that must be supported (e.g. Init, GC, digitalRead, digitalWrite)
  * naming of functions
* changes to wiringX.c/h (need to #include platform.h)
* Python bindings
  * need to add platform.c in setup.py
* Documentation updates required

CurlyMoo, 2015-03-03:
Are you willing to create that howto?

crazystick, 2015-03-03:
I haven't added support for a new platform from scratch, so I'm probably not best suited to creating it, hint: @garysims has :-)

I'm of course willing to contribute to it

CurlyMoo, 2015-03-03:
Can you check:
https://github.com/wiringX/wiringX/blob/master/CONTRIBUTING.md

crazystick, 2015-03-08:
I think this is fine

closed 2015-03-08 by crazystick