wiringX/wiringX #22: Kernel panic in xbian while using wiringX

skpacman, 2015-05-05:
I have a wiringX python script sending data to a LCD screen via the RPi2B's GPIO. The RPi2B is running the latest XBian. I was having trouble with random freezes, shutdowns, and kernel panics until I shut down the LCD python script from running at boot. Now the LCD is blank, but the system runs without issues.

**It's not my power supply.** It's a 5V 5A supply and was last measured pumping out 5.02V at full load (5A) and 4.99V at normal load (1.4A) with no fluctuation at a constant load. It never dropped below 4.95V, and never went above 5.05V during the test (2-hour load test)

*Could running a GPIO-heavy application like wiringX on XBian cause this or could there be something else in my script causing this?*

CurlyMoo, 2015-05-05:
I doubt it because i also run XBian with wiringX from pilight and have no issues.

skpacman, 2015-05-05:
It doesn't make sense then. I disabled my python script that uses wiringX and it's back to normal. Before disabling it, I couldn't go more than 24 hours without it crashing/panicking/etc. and since I disabled it, it's gone 4 days with no problems now.

When I say this script is GPIO-heavy, it's toggling an output as a data stream (bytes), about 1024 times every few seconds (8 bits X 128 columns) at a minimum (sometimes more), while 3 other outputs are just a few times per second. I'm not doing slow toggles of LED's or something like that.

The only other custom thing I have in the script is the psutil stuff. I couldn't find a suitable version so I had to build one that's not in XBian's repository. I'll look through my logs and see if there's anything useful.

If someone else can't reproduce this with a similar setup, it's gotta be something I'm doing, right?

closed 2015-05-05 by skpacman