wiringX/wiringX #23: How to use hardware interrupt in wiringX?

ypflll, 2015-05-20:
I want to use hardware interrupt on a radxa board. I read radxa.c in wiringX and find that in function "static int radxaISR(int pin, int mode)",  no user defined interrupt function be called.
However, in wiringPi  (see https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c), the function is defined like this: int wiringPiISR (int pin, int mode, void (*function)(void))--see a pointer of user defined function is imported, and in complementation of this function, an interruptHandler is called by:
pthread_create (&threadId, NULL, interruptHandler, &pin).  
But in radxa.c, I didn't see anywhere calling the "interruptHandler".
So, I am a little confused that  where the user interrupt handling function  is called? And how does radxaISR calls the interruptHandler?

CurlyMoo, 2015-05-20:
https://github.com/wiringX/wiringX/blob/master/examples/interrupt.c

ypflll, 2015-05-21:
Seems that I should check the "examples" folder first.
A little curious why wiringX changes the way to call the interrupt and does not align with wiringPi?

CurlyMoo, 2015-05-21:
Because i don't think a GPIO library should not force you into using a fixed threaded implementation. In my implementation you can choose yourself how you implementent the interrupt handler inside your threads.

ypflll, 2015-05-26:
Hi, I met a problem when testing the ''interrupt" example on a Rock-pro board.
I input a PWM wave in the GPIO 1, and it output "timeout" all the time (with no intervals, and no other information) in the terminal. And I detect no PWM in port 0--it seems that the program is death cycling in "*interrupt" .
So what's the problem? Have you tested the example on Rock?

CurlyMoo, 2015-05-26:
The radxa doesn't support interrupt edge BOTH.

ypflll, 2015-05-29:
Hi, I've solved my problem:
That's because for Radxa, a OS with kernal vision of 3.0.36 has a pin_base of 160, but wiringX does's considerates this. (Also, 3.0.36 doesn't support edge BOTH). I have fixed this bug.
For kernal vision 3.18.0, its pin_base is 0, and supports BOTH edge. WiringX runs well and I have tested interrupt on both kernal.
However, I find another problem:
The example will produce a false interrrupt for the first time. It means that, when there is no signal input, it will print "interrupt\n  timeout\n timeout\n....". So, where does the first "interrupt" come from?

CurlyMoo, 2015-05-29:
> That's because for Radxa, a OS with kernal vision of 3.0.36 has a pin_base of 160, but wiringX does's considerates this. (Also, 3.0.36 doesn't support edge BOTH). I have fixed this bug.
For kernal vision 3.18.0, its pin_base is 0, and supports BOTH edge. WiringX runs well and I have tested interrupt on both kernal.

Can you maybe update the radxa code so it will work on both kernel 3.0.36 and 3.18.0? And also create a check that the BOTH edge is not supported on 3.0.36 but is on 3.18.0?

> The example will produce a false interrrupt for the first time. It means that, when there is no signal input, it will print "interrupt\n timeout\n timeout\n....". So, where does the first "interrupt" come from?

Don't know. Can you check if this is fixed when you do this:
```
pinMode(0, OUTPUT);
digitalWrite(0, LOW);
```
So it sets the output pin to low before the interrupt is initialized?

ypflll, 2015-05-29:
I am still testing modules on RadxaBoard. I'll sort out my code later.
What do you mean by create a check that the BOTH edge is not supported on 3.0.36 but is on 3.18.0?
I have tested on both kernal. BOTH edge works well on 3.18.0. I won't take effort to make BOTH edge work on 3.0.36 cause Raxda will support 4.0 soon, I think.

CurlyMoo, 2015-05-29:
> What do you mean by create a check that the BOTH edge is not supported on 3.0.36 but is on 3.18.0? I have tested on both kernal.

Users on 3.0.36 should get a warning that the BOTH edge is not supported to prevent complains.

ypflll, 2015-06-04:
Hi, I've figured it out. 
Why there is a first "interrupt" when no input--it's because in wiringX, you call "x = poll(&polls, 1, ms)" before "(void)read(sysFds[pin], &c, 1)", and you'll get an interrupt for the first time. 
You should call "(void)read(sysFds[pin], &c, 1)" to clear the interrupt before poll.

CurlyMoo, 2015-06-04:
Ok, i will look into it. The second faulty wiringPi implementation.... I already fixed the first one a while ago.

CurlyMoo, 2015-11-02:
On a sidenote. I would like to change the LICENSE of wiringX from GPLv3 to MPLv2. Would that be ok by you?

closed 2016-05-14 by CurlyMoo

shrikantlahase, 2016-09-30:
hi @ypflll 
where can i get kernel 3.18 for radxa lite , i want to implement ISR in radxa lite board....?
As you u said previously , kernel 3.0.doesn't support BOTH_EDGE interrupt

Regards,
Shrikant