wiringX/wiringX #31: About add serial and PWM to wiringX.

ypflll, 2015-06-09:
Hi, CurlyMoo. I'm trying to add serial and pwm to radxaboard, and i am wondering why they are not added to wiringX, now that RaspberryPi and BananaPi have such interfaces?

When I tried to do so, I found a question:
In wiringX.c, functions are implemented by calling platform related function, like wiringXI2CRead, we use x = platform->I2CRead(fd).   If x==-1, we think that platform->I2CRead is failed. But how about the I2C really reads a -1 as data? This also happens in serial port.

CurlyMoo, 2015-06-09:
As far as i know, serial communication only happens in characters, so the -1 should never occur.
I don't know about i2c.

ypflll, 2015-06-10:
I've added serial for radxa and tested it. How about adding serial to wiringX?

CurlyMoo, 2015-06-10:
Yes

ypflll, 2015-06-10:
Hi, after sync with your recent two commits, wiringX can't make install on radxa:

/home/rock/wiringX/radxa.c:272:12: error: invalid storage class for function ‘rockchip_gpio_set_mux’
 static int rockchip_gpio_set_mux(unsigned int pin, unsigned int mux) {
            ^
/home/rock/wiringX/radxa.c:318:12: error: invalid storage class for function ‘changeOwner’
 static int changeOwner(char *file) {
...
What's wrong?

CurlyMoo, 2015-06-10:
Fixed

ypflll, 2015-06-10:
Can I PR directly with no sync from yours?

CurlyMoo, 2015-06-10:
Why

ypflll, 2015-06-10:
Cause I'm not well familiar how to rebase..
No bother, I'll do it.

ypflll, 2015-06-11:
Hi, CurlyMoo. I'm trying to add PWM to wiringX and I met a problem.
In wiringPi, function 'mmap' is used to get register address, like:
gpio = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_BASE) ;
Then, all gpio can be read or write by 'gpio'. The same to 'pwm', 'clk' and 'timer'.
But in wiringX for radxa, it's more complicated. You use this function:
map_reg(bank->reg_base, &bank->reg_mapped_base);
In map_reg, an 'address offset' is calculated before 'mmap' and added. And 'map_reg' is called for four times in radxaSetup. And 'iomux offsets' is calculate.
I am confused with this. Can you tell me how do you get the address for a gpio port? Why 'gpio' should be 'mmap' for four times and what are 'address offset' and 'iomux offsets'?

ypflll, 2015-06-11:
Can I get PWM register address like this?
pwm = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_PWM) ;
And what is the value of GPIO_PWM?

hipboi, 2015-06-11:
@ypflll You need to mmap the IO address before using them in the user space. For rockchip platform, because the bank base address are not continuous, we need to map for each bank.

ypflll, 2015-06-15:
Here the PWM register address:
http://lists.infradead.org/pipermail/linux-arm-kernel/2014-August/280671.html

ypflll, 2015-06-16:
Hi, CurlyMoo. It can't make again:
Linking C shared library libwiringX.so
/usr/bin/ld: CMakeFiles/sources.dir/src/raspberrypi.c.o: relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
What's the problem?

CurlyMoo, 2015-06-16:
`cmake .. -DCMAKE_C_FLAGS=-fPIC`

UEckhardt, 2016-01-01:
Will PWM also work on a BananaPi? If not, I could try to get PWM to work on a BPI-R1 board.

CurlyMoo, 2016-01-01:
Currently only radxa.

closed 2016-05-14 by CurlyMoo