wiringX/wiringX #48: make install for libwiringx

Quark-X10, 2015-10-09:
At the moment it is impossible to "make install" and turn on the +x bit for libwiringX, this problem may be solved by replacing FILES by LIBRARY when calling install() for the libs in CMakeFiles.txt:100 (I'm not a CMake expert).

CurlyMoo, 2015-10-09:
What platform are we talking about?

Quark-X10, 2015-10-09:
Oh sorry, I'm using Linux. I have tried with cmake 2.8 and 3.2

Quark-X10, 2015-10-09:
I'm doing tests on my debian laptop and on my gentoo HB and the behaviour is the same

CurlyMoo, 2015-10-09:
If you follow the installation steps you can just install the final deb or rpm.

Quark-X10, 2015-10-09:
Yeah i could, but I use gentoo on my HB so I don't want to make an extra dependency on dpkg and cpack where a simple "make install" can do the work.

CurlyMoo, 2015-10-09:
What files would need the executable bit?

Quark-X10, 2015-10-09:
the two libraries generated : libwiringX.so and libwringX.a

CurlyMoo, 2015-10-09:
Why do you want to `execute` them?

Quark-X10, 2015-10-09:
Oh ! Sorry i had a missing file problem with libwiringX.so when launching my program (which used the lib) and tought the problem was the missing +x but it got fixed. Sorry for your time

closed 2015-10-09 by Quark-X10