wiringX/wiringX #51: libwiringX.so: undefined reference to `i2c_smbus_xxx'

vitalogy, 2015-11-25:
Hi,

getting this with latest git:

```
In file included from /home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/radxa.c:33:0:
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/radxa.c: At top level:
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:23:14: warning: inline function 'i2c_smbus_write_word_data' declared but never defined
 inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:22:14: warning: inline function 'i2c_smbus_read_word_data' declared but never defined
 inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:21:14: warning: inline function 'i2c_smbus_write_byte_data' declared but never defined
 inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:20:14: warning: inline function 'i2c_smbus_read_byte_data' declared but never defined
 inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:19:14: warning: inline function 'i2c_smbus_write_byte' declared but never defined
 inline __s32 i2c_smbus_write_byte(int fd, int value);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:18:14: warning: inline function 'i2c_smbus_read_byte' declared but never defined
 inline __s32 i2c_smbus_read_byte(int fd);
              ^
/home/michael/workspace/pkgbuild/wiringx/src/wiringx-git/src/i2c-dev.h:17:14: warning: inline function 'i2c_smbus_access' declared but never defined
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^
[ 44%] Building C object CMakeFiles/sources.dir/src/i2c-dev.c.o
[ 44%] Built target sources
Scanning dependencies of target wiringx_shared
[ 50%] Linking C shared library libwiringX.so
[ 50%] Built target wiringx_shared
Scanning dependencies of target wiringx-blink
[ 55%] Building C object CMakeFiles/wiringx-blink.dir/examples/blink.c.o
[ 61%] Linking C executable wiringx-blink
libwiringX.so: undefined reference to `i2c_smbus_write_byte'
libwiringX.so: undefined reference to `i2c_smbus_read_byte'
libwiringX.so: undefined reference to `i2c_smbus_write_byte_data'
libwiringX.so: undefined reference to `i2c_smbus_write_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_byte_data'
collect2: error: ld returned 1 exit status
CMakeFiles/wiringx-blink.dir/build.make:93: recipe for target 'wiringx-blink' failed
make[2]: *** [wiringx-blink] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/wiringx-blink.dir/all' failed
make[1]: *** [CMakeFiles/wiringx-blink.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

CurlyMoo, 2015-11-26:
No issues here:
```
root@xbian /tmp # GIT_SSL_NO_VERIFY=true git clone https://github.com/wiringX/wiringX.git
Cloning into 'wiringX'...
remote: Counting objects: 689, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 689 (delta 2), reused 0 (delta 0), pack-reused 681
Receiving objects: 100% (689/689), 593.59 KiB | 0 bytes/s, done.
Resolving deltas: 100% (407/407), done.
Checking connectivity... done.
root@xbian /tmp # cd wiringX/
root@xbian /tmp/wiringX # cmake ^C
root@xbian /tmp/wiringX # mkdir build
root@xbian /tmp/wiringX # cd build/
root@xbian /tmp/wiringX/build # cmake ..
-- The C compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/wiringX/build
root@xbian /tmp/wiringX/build # make
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/wiringX/build
Scanning dependencies of target sources
[  9%] Building C object CMakeFiles/sources.dir/src/bananapi.c.o
[ 18%] Building C object CMakeFiles/sources.dir/src/ci20.c.o
[ 27%] Building C object CMakeFiles/sources.dir/src/hummingboard.c.o
[ 36%] Building C object CMakeFiles/sources.dir/src/i2c-dev.c.o
[ 45%] Building C object CMakeFiles/sources.dir/src/radxa.c.o
/tmp/wiringX/src/radxa.c:78:1: warning: useless storage class specifier in empty declaration
 };
 ^
/tmp/wiringX/src/radxa.c:109:1: warning: useless storage class specifier in empty declaration
 };
 ^
/tmp/wiringX/src/radxa.c:122:1: warning: useless storage class specifier in empty declaration
 };
 ^
/tmp/wiringX/src/radxa.c: In function ‘radxaSetup’:
/tmp/wiringX/src/radxa.c:51:25: warning: passing argument 1 of ‘map_reg’ makes pointer from integer without a cast
 #define RK30_AIN_BASE   0x2006c000
                         ^
/tmp/wiringX/src/radxa.c:684:16: note: in expansion of macro ‘RK30_AIN_BASE’
  ret = map_reg(RK30_AIN_BASE, &ain);
                ^
/tmp/wiringX/src/radxa.c:243:12: note: expected ‘void *’ but argument is of type ‘int’
 static int map_reg(void *reg, void **reg_mapped) {
            ^
/tmp/wiringX/src/radxa.c:684:31: warning: passing argument 2 of ‘map_reg’ from incompatible pointer type
  ret = map_reg(RK30_AIN_BASE, &ain);
                               ^
/tmp/wiringX/src/radxa.c:243:12: note: expected ‘void **’ but argument is of type ‘volatile void **’
 static int map_reg(void *reg, void **reg_mapped) {
            ^
[ 54%] Building C object CMakeFiles/sources.dir/src/raspberrypi.c.o
[ 63%] Building C object CMakeFiles/sources.dir/src/wiringX.c.o
[ 63%] Built target sources
Scanning dependencies of target wiringx_shared
Linking C shared library libwiringX.so
[ 63%] Built target wiringx_shared
Scanning dependencies of target wiringx-blink
[ 72%] Building C object CMakeFiles/wiringx-blink.dir/examples/blink.c.o
Linking C executable wiringx-blink
[ 72%] Built target wiringx-blink
Scanning dependencies of target wiringx-i2c-spi
[ 81%] Building C object CMakeFiles/wiringx-i2c-spi.dir/examples/i2c_spi.c.o
Linking C executable wiringx-i2c-spi
[ 81%] Built target wiringx-i2c-spi
Scanning dependencies of target wiringx-interrupt
[ 90%] Building C object CMakeFiles/wiringx-interrupt.dir/examples/interrupt.c.o
Linking C executable wiringx-interrupt
[ 90%] Built target wiringx-interrupt
Scanning dependencies of target wiringx-read
[100%] Building C object CMakeFiles/wiringx-read.dir/examples/read.c.o
Linking C executable wiringx-read
[100%] Built target wiringx-read
Scanning dependencies of target wiringx_static
Linking C static library libwiringX.a
[100%] Built target wiringx_static
root@xbian /tmp/wiringX/build #
```

vitalogy, 2015-11-27:
building here on a cubietruck on up-to-date Arch-Linux with  gcc-5.2.0-2 / cmake-3.4.0-2

```
michael@cubietruck git $ LANG=C git clone https://github.com/wiringX/wiringX.git
Cloning into 'wiringX'...
remote: Counting objects: 689, done.
remote: Total 689 (delta 0), reused 0 (delta 0), pack-reused 689
Receiving objects: 100% (689/689), 591.38 KiB | 598.00 KiB/s, done.
Resolving deltas: 100% (410/410), done.
Checking connectivity... done.
michael@cubietruck git $ cd wiringX/
michael@cubietruck wiringX $ cmake .
-- The C compiler identification is GNU 5.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/michael/workspace/git/wiringX
michael@cubietruck wiringX $ LANG=C make
Scanning dependencies of target sources
[  5%] Building C object CMakeFiles/sources.dir/src/bananapi.c.o
In file included from /home/michael/workspace/git/wiringX/src/bananapi.c:30:0:
/home/michael/workspace/git/wiringX/src/i2c-dev.h:23:14: warning: inline function 'i2c_smbus_write_word_data' declared but never defined
 inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:22:14: warning: inline function 'i2c_smbus_read_word_data' declared but never defined
 inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:21:14: warning: inline function 'i2c_smbus_write_byte_data' declared but never defined
 inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:20:14: warning: inline function 'i2c_smbus_read_byte_data' declared but never defined
 inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:19:14: warning: inline function 'i2c_smbus_write_byte' declared but never defined
 inline __s32 i2c_smbus_write_byte(int fd, int value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:18:14: warning: inline function 'i2c_smbus_read_byte' declared but never defined
 inline __s32 i2c_smbus_read_byte(int fd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:17:14: warning: inline function 'i2c_smbus_access' declared but never defined
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^
[ 11%] Building C object CMakeFiles/sources.dir/src/raspberrypi.c.o
In file included from /home/michael/workspace/git/wiringX/src/raspberrypi.c:33:0:
/home/michael/workspace/git/wiringX/src/i2c-dev.h:23:14: warning: inline function 'i2c_smbus_write_word_data' declared but never defined
 inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:22:14: warning: inline function 'i2c_smbus_read_word_data' declared but never defined
 inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:21:14: warning: inline function 'i2c_smbus_write_byte_data' declared but never defined
 inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:20:14: warning: inline function 'i2c_smbus_read_byte_data' declared but never defined
 inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:19:14: warning: inline function 'i2c_smbus_write_byte' declared but never defined
 inline __s32 i2c_smbus_write_byte(int fd, int value);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:18:14: warning: inline function 'i2c_smbus_read_byte' declared but never defined
 inline __s32 i2c_smbus_read_byte(int fd);
              ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:17:14: warning: inline function 'i2c_smbus_access' declared but never defined
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^
[... 94 lines cut ...]
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^
[ 41%] Building C object CMakeFiles/sources.dir/src/i2c-dev.c.o
[ 41%] Built target sources
Scanning dependencies of target wiringx_shared
[ 47%] Linking C shared library libwiringX.so
[ 47%] Built target wiringx_shared
Scanning dependencies of target wiringx-blink
[ 52%] Building C object CMakeFiles/wiringx-blink.dir/examples/blink.c.o
[ 58%] Linking C executable wiringx-blink
libwiringX.so: undefined reference to `i2c_smbus_write_byte'
libwiringX.so: undefined reference to `i2c_smbus_read_byte'
libwiringX.so: undefined reference to `i2c_smbus_write_byte_data'
libwiringX.so: undefined reference to `i2c_smbus_write_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_byte_data'
collect2: error: ld returned 1 exit status
CMakeFiles/wiringx-blink.dir/build.make:93: recipe for target 'wiringx-blink' failed
make[2]: *** [wiringx-blink] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/wiringx-blink.dir/all' failed
make[1]: *** [CMakeFiles/wiringx-blink.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
michael@cubietruck wiringX $ 
```

CurlyMoo, 2015-11-27:
I have no clue since i can't reproduce...

vitalogy, 2015-11-27:
i have searched and read ... and got it build with ```extern inline``` in i2c-dev.c
but i don't know, if this the right way

CurlyMoo, 2015-11-27:
Can you try again with the latest commit?

vitalogy, 2015-11-27:
```
In file included from /home/michael/workspace/git/wiringX/src/radxa.c:33:0:
/home/michael/workspace/git/wiringX/src/radxa.c: Auf höchster Ebene:
/home/michael/workspace/git/wiringX/src/i2c-dev.h:23:21: Warnung: Inline-Funktion »i2c_smbus_write_word_data« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:22:21: Warnung: Inline-Funktion »i2c_smbus_read_word_data« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:21:21: Warnung: Inline-Funktion »i2c_smbus_write_byte_data« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:20:21: Warnung: Inline-Funktion »i2c_smbus_read_byte_data« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:19:21: Warnung: Inline-Funktion »i2c_smbus_write_byte« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_write_byte(int fd, int value);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:18:21: Warnung: Inline-Funktion »i2c_smbus_read_byte« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_read_byte(int fd);
                     ^
/home/michael/workspace/git/wiringX/src/i2c-dev.h:17:21: Warnung: Inline-Funktion »i2c_smbus_access« deklariert, aber nirgendwo definiert
 extern inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
                     ^
[ 41%] Building C object CMakeFiles/sources.dir/src/i2c-dev.c.o
[ 41%] Built target sources
Scanning dependencies of target wiringx_shared
[ 47%] Linking C shared library libwiringX.so
[ 47%] Built target wiringx_shared
Scanning dependencies of target wiringx-blink
[ 52%] Building C object CMakeFiles/wiringx-blink.dir/examples/blink.c.o
[ 58%] Linking C executable wiringx-blink
libwiringX.so: undefined reference to `i2c_smbus_write_byte'
libwiringX.so: undefined reference to `i2c_smbus_read_byte'
libwiringX.so: undefined reference to `i2c_smbus_write_byte_data'
libwiringX.so: undefined reference to `i2c_smbus_write_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_word_data'
libwiringX.so: undefined reference to `i2c_smbus_read_byte_data'
```

CurlyMoo, 2015-11-27:
Please provide your fix.

vitalogy, 2015-11-27:
i have set ```extern inine``` in the source file (i2c-dev.c)

found this here
http://stackoverflow.com/questions/6312597/is-inline-without-static-or-extern-ever-useful-in-c99

CurlyMoo, 2015-11-27:
I meant in a PR.

vitalogy, 2015-11-27:
#52

vitalogy, 2015-11-27:
is this error related to gcc-5, because the default mode for C in gcc-5 is now -std=gnu11 instead of -std=gnu89?

CurlyMoo, 2015-11-27:
Don't know

vitalogy, 2015-11-28:
#53
no warnings or error at compile time

closed 2015-11-28 by CurlyMoo