wiringX/wiringX #54: WiringX on (other) MIPS devices

466, 2015-12-23:
Like the MIPS CI20 Creator, there are a lot MIPS based platforms out there
(e.g. ar71xx; https://dev.openwrt.org/wiki/platforms ).
I was wondering what steps needs to be taken in order to use WiringX on these platforms.

I understand for every hardware, new source files do have to get created
e.g.
ar71xx.c
ar71xx.h
which enables a platform check via /proc/cpuinfo (or any other way)
and allows GPIO usage, preferable using direct access to it's registers, via /dev/mem.

In the first place, i downloaded AR9331's and CI20's datasheets
(AR9931: http://www.datasheet4u.com/download.php?id=743003
CI20/JZ4780 : http://mipscreator.imgtec.com/CI20/hardware/soc/JZ4780_PM.pdf) in order to
a) find out what the registers, ci20.c refers to, are originally called
b) what their corresponding registers of AR9331 might look like.

But as this is quite new to me, all these different registers, names and what ever you find, I hope there is someone here who is way more into this topic than i am :P ....

CurlyMoo, 2015-12-23:
That's indeed basicly it. Figuring out the registers requires most work.

466, 2015-12-28:
Ok, thanks for your reply.

I've had little time during Christmas, now, back to topic:

I did some changes to the code (ci20), saved it as ap121.c and ap121.h +changed wiringX.c accordingly:
############################
#include "hummingboard.h"
#include "raspberrypi.h"
#include "bananapi.h"
#include "ci20.h"
#include "ap121.h"
############################

############################
			hummingboardInit();
			raspberrypiInit();
			bananapiInit();
			ci20Init();
			ap121Init();
############################
I was able to compile and install it.
But it does not work, due to wiringX not being able to find supported hardware.
Due to the logfiles, it seems to me, that wiringX does only test for huminboard->identfy and bananapi->identify,
not for ci20 or ap121.

CONTRIBUTING.md says 
15. Add your new module to wiringX.c and wiringX.h.

Is this up-to-date?
If so, where do i have to add the module within wiringX.h?

OR did i simply miss something?

CurlyMoo, 2016-05-14:
You can check the new rewrite branch. That should help you port the other MIPS devices much more easily.

closed 2022-10-28 by CurlyMoo