wiringX/wiringX #55: Support for PCDuino3 board

rkakrik, 2016-01-04:
PCDuino3 is similar to Banana Pi board, but with a different arrangement of pins. Some pins are worked with wiringX: pcduino gpio 0, 1, 7, 9, 10, 11, 12, 13 match with gpio 0, 2, 7, 17, 10, 12, 13, 14 in wiringX, respectively.
It is possible to add full support for pcduino? By default in pcduino's stock ubuntu all pins listed in sysfs, but I use archlinux with vanilla kernel and wiringX. Maybe there is a way to get the address for /dev/mem from ubuntu?

CurlyMoo, 2016-01-04:
Can you post the content of /proc/cpuinfo?

rkakrik, 2016-01-08:
Processor	: ARMv7 Processor rev 4 (v7l)
processor	: 0
BogoMIPS	: 1816.98

processor	: 1
BogoMIPS	: 1823.53

Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 4

Hardware	: sun7i
Revision	: 0000
Serial		: 0000000000000000

CurlyMoo, 2016-01-08:
Thanks. I have some ideas on this, but it's not high on my priority list.

rkakrik, 2016-01-09:
If needed, I can make ssh access

CurlyMoo, 2016-05-14:
You can check the new rewrite branch. That should help you port the PCDuino3 much more easily.

closed 2019-11-26 by CurlyMoo