wiringX/wiringX #56: Disable "running on"

icaruseffect, 2016-01-23:
Hi,

Use Case: I am running using wiringX to interface a DHT11 on a bananapi using this [fork] and logging to monit using the [munin-dht] plugin

Problem: Every time i call the library from munin-dht to read a value, I get  "running on bananapi" as first result.  This is invoked in src/wiringX.c line 700 with :
> wiringXLog(LOG_DEBUG, "running on a %s", platform->name);
This results in monit not eating the output, because of the 

Question: Is it possible, to add a toggle for a debug-mode and wrap output which is not relevant outside development in it?

[fork]: https://github.com/icaruseffect/Adafruit_Python_DHT
[munin-dht]:https://github.com/agx/munin-dht

CurlyMoo, 2016-01-23:
You can also overwrite the logprintf function.

icaruseffect, 2016-01-23:
For now I just commented it out.  Your idea seems to be more elegant. 
Is there a switch accessible from the python library to alter the behavior of debugging messages?

closed 2016-05-14 by CurlyMoo