wiringX/wiringX #69: read() failed in the soc_wait_for_interrupt() function

bkrepo, 2016-04-14:
read() function always returns 0 value in the soc_wait_for_interrupt().
- https://github.com/wiringX/wiringX/blob/rewrite/src/soc/soc.c#L235

This code affects the read() operation.
- https://github.com/wiringX/wiringX/blob/rewrite/src/soc/soc.c#L215-L220

CurlyMoo, 2016-04-14:
Not sure what the issue?

bkrepo, 2016-04-18:
```
diff --git a/src/soc/soc.c b/src/soc/soc.c
index a338e57..cf1babd 100644
--- a/src/soc/soc.c
+++ b/src/soc/soc.c
@@ -233,6 +233,7 @@ int soc_wait_for_interrupt(struct soc_t *soc, int fd, int ms) {
        polls.events = POLLPRI;
 
        x = read(fd, &c, 1);
+       wiringXLog(LOG_DEBUG, "%s: read() ret = %d", __func__, x);
        if(x != 1) {
                return -1;
        }
```
and
* int_test.c
```
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <wiringX.h>

int main(int argc, char *argv[]) {
	int gpio_in = 0;
	int i = 0;
	int ret = 0;

	// expect 2 arguments => argc must be 3
	if(argc != 2) {
		fprintf(stderr, "%s <GPIO number>\n", argv[0]);
		return -1;
	}

	gpio_in = atoi(argv[1]);

	if(wiringXSetup("odroidc1", NULL) == -1) {
		wiringXGC();
		return -1;
	}

	if(wiringXValidGPIO(gpio_in) != 0) {
		printf("%s: Invalid GPIO %d for input (interrupt)\n", argv[0], gpio_in);
		wiringXGC();
		return -1;
	}

	if((wiringXISR(gpio_in, ISR_MODE_BOTH)) != 0) {
		wiringXGC();
		return -1;
	}

	ret = waitForInterrupt(gpio_in, 3000) > 0);
	printf("waitForInterrupt() ret = %d\n", ret);

	wiringXGC();

	return 0;
}
```
Outout:
```
root@odroid64:~# ./a.out 7
DEBUG: soc_wait_for_interrupt: read() ret = 0
waitForInterrupt() ret = -1
```
I have a question.
What does the https://github.com/wiringX/wiringX/blob/rewrite/src/soc/soc.c#L215-L220 routine do? Is it  clearing the pending interrupts?

CurlyMoo, 2016-04-18:
> Is it clearing the pending interrupts?

Yes

bkrepo, 2016-04-19:
The problem is in ODROID-C1/C2 that the file pointer does not reset automatically after processing the pending interrupts. One of the solution is using lseek() after interrupt initializing. 
```
diff --git a/src/soc/soc.c b/src/soc/soc.c
index a338e57..87bceb3 100644
--- a/src/soc/soc.c
+++ b/src/soc/soc.c
@@ -219,6 +219,7 @@ int soc_sysfs_gpio_reset_value(struct soc_t *soc, char *path) {
                                continue;
                        }
                }
+               lseek(fd, 0, SEEK_SET); 
        }
 
        return fd;
```
Is this acceptable?

CurlyMoo, 2016-04-19:
Yes.

closed 2016-04-20 by bkrepo

CurlyMoo, 2016-04-20:
I'm currently testing the Odroids and changing some thing here and there. Can you checkup on the changes i'm doing...

bkrepo, 2016-04-20:
There are nothing additional changes for ODROID boards now. (sync)
If some works are needed for ODROIDs, please make the issue and let me know it. I will work it in that case.

CurlyMoo, 2016-04-20:
For now i added the i2c pins as usable gpio and changed the pinmapping a bit. Did not test the interrupts yet.

bkrepo, 2016-04-20:
Ah~you are right. I2C pins are usable gpio.

I did not add there pins in map[] when wrote odroidc1.c source file. Because, I just made map[] array the same as wiringPi library due to the compatibility and user experience.

https://github.com/hardkernel/wiringPi/blob/master/wiringPi/wiringPi.c#L942-L950

There pins is possible for digital in/out. If you want to use there pins, please add the gpio number to padding index in the map[] array. But, I recommend not to change the previous gpio number in the map[] array due to compatibility issue.

CurlyMoo, 2016-04-20:
You can view my commit to see what changed. I updated the pin map so all platforms have the same pin layouts, and sadly the wiringPi map has become leading in this. 

The problem that might arise are in the interrupt mapping, but then we just need to add an interrupt map as well.

The I2C are also usable GPIO on the Rasberry Pi, so that's why i added them. I noticed that i couldn't use the UART pins as GPIO while i can on the other platforms.