wiringX/wiringX #72: A10-OLinuXino-LIME support problem

ajvuik, 2016-04-22:
Hello all,
I am currently working on support for the A10-OlinuXino platform from Olimex. It is almost complete, but during testing I came across a problem. I need to manually export and set the direction of a pin for WiringX to work.
I'll try do dive into the problem a bit deeper, but, perhaps someone here can give me some pointers on where to look for.

I'm not getting any error messages, or anything. Just the IO won't go on or off if I don't set the direction manualy in bash.

CurlyMoo, 2016-04-22:
Did you already put the code online in your fork?

ajvuik, 2016-04-22:
Yes, it's on-line in my fork.

CurlyMoo, 2016-04-22:
And where is that so we don't have so search for it ourselves.

ajvuik, 2016-04-22:
In /src/platform/Olimex are the files that I added. It uses the Allwinner A10 SOC. 
I also added this directory in the CMakeLists.txt so it will be included when building WiringX.
WiringX.c had been modified to add the .h file and init the platform.

CurlyMoo, 2016-04-22:
Where is your fork?

ajvuik, 2016-04-22:
https://github.com/ajvuik/wiringX

CurlyMoo, 2016-04-22:
How do you know you created the right map when wiringx-blink doesn't work?

ajvuik, 2016-04-22:
You mean the GPIO map?
Because when I export and set the direction manually and I run the blink program, it works.

... on hinsight, there is 1 zero to much at the start. This was to try if I made a mistake with the mapping, but as far as I can see, there isn't. So the 1st mapping can be omitted.

CurlyMoo, 2016-04-22:
You can't rely on the userspace numbers. You need to map the pins by (bulk) trying the memory areas from the A10 module.

ajvuik, 2016-04-22:
No, I cross reference the userspace with the memory mapping to check if I have the right GPIO.

What I did is: I used the BananaPi 1 platform files and used them as a template for the new platform files.

CurlyMoo, 2016-04-22:
That what i said, the memory map number can be totally different from the userspace number. That's why the solidrun platforms have a separate gpio map and a irq map.

closed 2019-11-26 by CurlyMoo