wiringX/wiringX #81: Possible memory address typos in rewrite branch soc/alwinner/a10.c

Soulsurfering, 2016-05-29:
I think, there is an address typo in the memory mapping of Allwinner A10 file which I have found, when doing the Allwinner R8 mapping. The correct values are:
 { "PB8", 0, { 0x**28**, 0 }, { 0x34, 8 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
 ... 
 { "PG0", 0, { 0x**D8**, 0 }, { 0xE8, 0 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
 ...

CurlyMoo, 2016-05-29:
Do you agree with the following:
```
-  { "PB16", 0, { 0x32, 0 }, { 0x34, 16 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB17", 0, { 0x32, 4 }, { 0x34, 17 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB18", 0, { 0x32, 8 }, { 0x34, 18 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB19", 0, { 0x32, 12 }, { 0x34, 19 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB20", 0, { 0x32, 16 }, { 0x34, 20 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB21", 0, { 0x32, 20 }, { 0x34, 21 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB22", 0, { 0x32, 24 }, { 0x34, 22 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PB23", 0, { 0x32, 28 }, { 0x34, 23 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB16", 0, { 0x2C, 0 }, { 0x34, 16 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB17", 0, { 0x2C, 4 }, { 0x34, 17 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB18", 0, { 0x2C, 8 }, { 0x34, 18 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB19", 0, { 0x2C, 12 }, { 0x34, 19 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB20", 0, { 0x2C, 16 }, { 0x34, 20 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB21", 0, { 0x2C, 20 }, { 0x34, 21 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB22", 0, { 0x2C, 24 }, { 0x34, 22 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PB23", 0, { 0x2C, 28 }, { 0x34, 23 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
```

and
```
-  { "PG0", 0, { 0xDC, 0 }, { 0xE8, 0 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG1", 0, { 0xDC, 4 }, { 0xE8, 1 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG2", 0, { 0xDC, 8 }, { 0xE8, 2 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG3", 0, { 0xDC, 12 }, { 0xE8, 3 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG4", 0, { 0xDC, 16 }, { 0xE8, 4 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG5", 0, { 0xDC, 20 }, { 0xE8, 5 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG6", 0, { 0xDC, 24 }, { 0xE8, 6 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG7", 0, { 0xDC, 28 }, { 0xE8, 7 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG8", 0, { 0xE0, 0 }, { 0xE8, 8 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG9", 0, { 0xE0, 4 }, { 0xE8, 9 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG10", 0, { 0xE0, 8 }, { 0xE8, 10 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
-  { "PG11", 0, { 0xE0, 12 }, { 0xE8, 11 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG0", 0, { 0xD8, 0 }, { 0xE8, 0 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG1", 0, { 0xD8, 4 }, { 0xE8, 1 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG2", 0, { 0xD8, 8 }, { 0xE8, 2 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG3", 0, { 0xD8, 12 }, { 0xE8, 3 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG4", 0, { 0xD8, 16 }, { 0xE8, 4 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG5", 0, { 0xD8, 20 }, { 0xE8, 5 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG6", 0, { 0xD8, 24 }, { 0xE8, 6 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG7", 0, { 0xD8, 28 }, { 0xE8, 7 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG8", 0, { 0xDC, 0 }, { 0xE8, 8 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG9", 0, { 0xDC, 4 }, { 0xE8, 9 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG10", 0, { 0xDC, 8 }, { 0xE8, 10 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
+  { "PG11", 0, { 0xDC, 12 }, { 0xE8, 11 }, FUNCTION_DIGITAL, PINMODE_NOT_SET, 0 },
```

Soulsurfering, 2016-05-29:
Yes, I think that now correct.

kgbkgb, 2016-07-26:
I am planning to use the OrangePi PC computer, Armbian 5.14,  to implement a system with ADC, DACs and a counter to replace an outdated MSDOS system I have built
I need to use an interrupt input to make a counter, exactly as in the "isr.c" example in WiringPi.. This works fine with WiringPi on a Rasberrypi 2. I tried this using WiringOP on the OrangePi PC (H3 cpu) but it is flaky. 
WiringX works better on the OrangePi but the interrupt support seems to only allow a "wait for interrupt" routine. 
WiringPi also has a "link to user supplied interrupt service routine upon interrupt"
      int wiringPiISR (int pin, int mode, void (*function)(void))
I need to use this type of interrupt routine to avoid the polling or blocking read in the "wait for interrupt".
Is there some problem preventing this implementation in  WiringX?
I am not a system programmer, but should i be able to implement this function by "pasting" across the int wiringPiISR function from WiringPi and attempting to fix variable names etc? 
(Linux User-land interrupts are OK as the frequency is <300Hz and not response-time critical)

closed 2019-11-26 by CurlyMoo