wiringX/wiringX #82: Banana Pi and Banana Pi / Lamobo R1 GPIO definitions

alexschomb, 2016-06-25:
The GPIO pin header is said to be identical to the original Banana Pi, but GPIO J11 is called GPIO J13. I found that the GPIO numbering on the wiringX homepage might be wrong, especially concerning J11 (J13) and J12. Shouldn't BCM 28, 29, 30, 31 be wiringX numbers 17, 18, 19, 20? ([see here](https://github.com/wiringX/wiringX/blob/00ef02a41b46dd626bf6b66309b0ec140d779075/src/bananapi.c#L96)) What about J12_P03 and J12_P04? They have the same wiringX number as CON3_P08 and CON3_P09. Can you help?

You can find the GPIO definitions of the Banana Pi / Lamobo R1 here: https://bananapi.gitbooks.io/bpi-r1/content/en/bpi-r1_gpio_pin_define.html

CurlyMoo, 2016-06-25:
Have you checked the new wiringX rewrite code?

alexschomb, 2016-06-26:
I figured, because I'm using the ''rewrite'' branch of ''pilight'' which you thankfully suggested to me in the forums earlier.

CurlyMoo, 2016-06-26:
I mean, did you check the pin numbering in the rewrite source. It's also documentated there (for now):
https://github.com/wiringX/wiringX/blob/rewrite/src/platform/lemaker/bananapi1.c#L26-L45

alexschomb, 2016-06-26:
Sorry, you're right. Problem solved 👍 

I found that my problems resulted in a combination of wrong wiringX & GPIO number (calculation) and not exporting the correct GPIOs.

Thanks for your help, again!

closed 2016-06-26 by alexschomb