wiringX/wiringX #83: Interrupt service routine with callback

kgbkgb, 2016-07-26:
I am planning to use the OrangePi PC computer, Armbian 5.14, to implement a system with ADC, DACs and a counter to replace an outdated MSDOS system I have built
I need to use an interrupt input to make a counter, exactly as in the "isr.c" example in WiringPi.. This works fine with WiringPi on a Rasberrypi 2. I tried this using WiringOP on the OrangePi PC (H3 cpu) but it is flaky.
WiringX works better on the OrangePi but the interrupt support seems to only allow a "wait for interrupt" routine.
WiringPi also has a "link to user supplied interrupt service routine upon interrupt"
int wiringPiISR (int pin, int mode, void (*function)(void))
I need to use this type of interrupt routine to avoid the polling or blocking read in the "wait for interrupt".
Is there some problem preventing this implementation in WiringX?
I am not a system programmer, but should i be able to implement this function by "pasting" across the int wiringPiISR function from WiringPi and attempting to fix variable names etc?
(Linux User-land interrupts are OK as the frequency is <300Hz and not response-time critical)

CurlyMoo, 2016-07-27:
Can you post the code of your current implementation (interrupt part)?

kgbkgb, 2016-07-28:
This code which I mentioned is straight from WiringPi, file wiringPi.c.
Although this is present in the wiringOP conversion, it does not work - it compiles but the counter example program isr.c gets no counts. A wait-for-interrupt example on the same header pin can detect interrupts. The wiringOP version is rather confusing with Banana Pi modifications also.

The wiringOP package is here   [https://github.com/zhaolei/WiringOP](url)
I also include the example file isr.c from WiringPi, which uses this interrupt routine to provide a counter of interrupt events.

====  extract from wiringPi.c=======
`int wiringPiISR (int pin, int mode, void (*function)(void))
{
  pthread_t threadId ;
  const char *modeS ;
  char fName   [64] ;
  char  pinS [8] ;
  pid_t pid ;
  int   count, i ;
  char  c ;
  int   bcmGpioPin ;

  if ((pin < 0) || (pin > 63))
    return wiringPiFailure (WPI_FATAL, "wiringPiISR: pin must be 0-63 (%d)\n", pin) ;

  /**/ if (wiringPiMode == WPI_MODE_UNINITIALISED)
    return wiringPiFailure (WPI_FATAL, "wiringPiISR: wiringPi has not been initialised. Unable to continue.\n") ;
  else if (wiringPiMode == WPI_MODE_PINS)
    bcmGpioPin = pinToGpio [pin] ;
  else if (wiringPiMode == WPI_MODE_PHYS)
    bcmGpioPin = physToGpio [pin] ;
  else
    bcmGpioPin = pin ;
// Now export the pin and set the right edge
//	We're going to use the gpio program to do this, so it assumes
//	a full installation of wiringPi. It's a bit 'clunky', but it
//	is a way that will work when we're running in "Sys" mode, as
//	a non-root user. (without sudo)

  if (mode != INT_EDGE_SETUP)
  {
    /**/ if (mode == INT_EDGE_FALLING)
      modeS = "falling" ;
    else if (mode == INT_EDGE_RISING)
      modeS = "rising" ;
    else
      modeS = "both" ;

    sprintf (pinS, "%d", bcmGpioPin) ;

    if ((pid = fork ()) < 0)	// Fail
      return wiringPiFailure (WPI_FATAL, "wiringPiISR: fork failed: %s\n", strerror (errno)) ;

    if (pid == 0)	// Child, exec
    {
      /**/ if (access ("/usr/local/bin/gpio", X_OK) == 0)
      {
	execl ("/usr/local/bin/gpio", "gpio", "edge", pinS, modeS, (char *)NULL) ;
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: execl failed: %s\n", strerror (errno)) ;
      }
      else if (access ("/usr/bin/gpio", X_OK) == 0)
      {
	execl ("/usr/bin/gpio", "gpio", "edge", pinS, modeS, (char *)NULL) ;
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: execl failed: %s\n", strerror (errno)) ;
      }
      else
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: Can't find gpio program\n") ;
    }
    else		// Parent, wait
      wait (NULL) ;
  }

// Now pre-open the /sys/class node - but it may already be open if
//	we are in Sys mode...
[... 111 lines cut ...]

  for (;;)
  {
    gotOne = 0 ;
    printf ("Waiting ... ") ; fflush (stdout) ;

    for (;;)
    {
      for (pin = 0 ; pin < 8 ; ++pin)
      {
	if (globalCounter [pin] != myCounter [pin])
	{
	  printf (" Int on pin %d: Counter: %5d\n", pin, globalCounter [pin]) ;
	  myCounter [pin] = globalCounter [pin] ;
	  ++gotOne ;
	}
      }
      if (gotOne != 0)
	break ;
    }
  }

  return 0 ;
}
`

CurlyMoo, 2016-07-28:
I of course know what wiringPi implemented, but i want to know how you used the function in your own code.

kgbkgb, 2016-07-28:
Sorry CurlyMoo. I have not started coding this new project yet. I am still at the hardware feasibility point and without an interrupt service feature the project is not feasible on an OrangePi - i would have to use a genuine raspberryPi.  I did use an interrupt vector routine in my 1990 equipment, but that was TurboC on MSDOS and I am trying to escape from that evil environment!  ;-)
All i need is the isr.c  example code capability: a global counter of interrupt events. Events come from a detector and amplifier, with pre-conditioning  in CMOS hardware to prevent overlapping events etc. That counter will be read at regular intervals using the system timer somehow.

CurlyMoo, 2016-07-28:
Just start using wiringX and when you get stuck because i implemented the interrupt callback differently, come back again and i'll help you.

kgbkgb, 2016-07-29:
I dont understand how to start coding without having an interrupt function which has a callback to a user supplied interrupt service routine. Have I missed something here? In WiringX i can only find a "waitforInterrupt" function. This blocks awaiting an interrupt doesn't it? But I need the code to continue graphing the data and controlling the DACs and reading the ADC while interrupts are being counted by the ISR routine without blocking. I think i need to use the    wiringPiISR (BUTTON_PIN, INT_EDGE_FALLING, &myInterrupt)    type of function as in WiringPi, where the &myinterrupt function increments the interrupt counter and returns control to the main code while awaiting the next interrupts.
Are you suggesting there is a way to use waitforinterrupt () without needing to run a separate thread?

CurlyMoo, 2016-07-29:
> Are you suggesting there is a way to use waitforinterrupt () without needing to run a separate thread?

No there isn't but have you actually looked under the hood of the wiringPiISR:
```
    pthread_create (&threadId, NULL, interruptHandler, NULL) ;
```

That's why it's a stupid implementation because they force you to use their threaded implementation instead of your own.

kgbkgb, 2016-08-02:
I am having major issues trying to get interrupts to work on the orangepi-PC.
In wiringX file orangepi.c, there is the following tabulation:
static int edge [64] =
{
  -1, -1, -1, -1, 4, -1, -1, 7,   //support the INT
  8, 9, 10, 11, -1,-1, 14, 15,
  -1, 17, -1, -1, -1, -1, 22, 23,
  24, 25, -1, 27, -1, -1, -1, -1,      ....etc.
I tested all the GPIO pins on this list, but most of them fail with:
orangepi: error while calling isr
these GPIO# pins are 11, 14, 0, 3, 4, 5, 6, 2 - they all give this fail error.
Only 6 pins are accepted by the wiringx-interrupt 5 x example/test program:
these are GPIO#  7, 10, 13, 12, 15, 16
But none detected the interrupt trigger I used (which I tested on a raspberry pi 2, and which worked properly)
AT the end of these tests i found the following GPIO files in /sys/class/gpio:  10, 198, 199, 200, 201, 68, 7, 8, 9

To cross check, I used the wiringOP package. (yes i know about the library conflict between these packages and removed the conflicting libraries for each test). I think there is a major issue with pin number mappings and only GPIO# 0, 7, 14 work for interrupt input, but these fail to trigger on an input signal. I suspect the mode selections for edge triggering are badly broken, probably as this package was converted first to a bananapi with A20 cpu, then to orangepi with H3 cpu.

I used a RaspberryPi model 2 with wiringPi and it worked correctly and confirmed that my interrupt input circuit is OK. 

In my early wiringX tests I thought i did get edge triggering working with the wiringx-wfi example program, but i cannot repeat this test. And the pin number mapping confusion means only one or 2 pins are available as I am using I2C and also SPI devices.
At present I will probably change to use Raspberry hardware as it and wiringPi work fine. I would still like to fix the int issues on wiringx on orangepi-PC, but right now i am too frustrated.

CurlyMoo, 2016-08-02:
You could ask OrangePi to create the wiringX module themselves. Other companies did so as well.

closed 2019-11-26 by CurlyMoo