wiringX/wiringX #84: wiringXISR is not working in python bindings

guysoft, 2016-09-08:
Hey,
So I tried running the interrupt example in the rewrite branch, but I get:
(after changing ``from wiringX.gpio import gpio`` to ``import wiringX.gpio as gpio``

``` 
Traceback (most recent call last):
  File "/home/pi/scripts/button_control.py", line 31, in <module>
    gpio.wiringXISR(gpio.PIN1, gpio.ISR_EDGE_BOTH);
AttributeError: 'module' object has no attribute 'wiringXISR'
```

It seems like wiringXISR is commented out:
https://github.com/wiringX/wiringX/blob/rewrite/python/wiringX/wiringx.c#L344

Is this feature disabled?
Is there a way to make a button interrupt using wiringx?

Thanks

CurlyMoo, 2016-09-08:
I don't have enough knowledge to enable wiringXISR in python, that's why. I hoped there would be someone to fix it for us.

guysoft, 2016-09-09:
Does it work in C?

guysoft, 2019-11-27:
Hey, the issue is not solved. I have moved on and can't help much, but closing it you might at least add "wontfix" or "rebopen if someone wants to contribute code"

CurlyMoo, 2019-11-27:
Ok, done