wiringX/wiringX #87: ci20 + wiringX LED blink example = gpio.setup()  IOError: [Errno 2] No such file or directory

soocki, 2017-01-16:
I have installed wiringX (both python and c versions) on the ci20, however I can't make use of the python's gpio.setup() function since it fails with the error

gpio.setup() IOError: [Errno 2] No such file or directory

My understanding was that it should work sort of out-of-the-box after installation, it doesn't seems to be the case. Can anyone point me in direction of solving this silly issue?

soocki, 2017-01-17:
The CI20 is not yet included in the rewrite branch, is it?

CurlyMoo, 2017-01-17:
Nope, the new code follows a different approach. I don't own a CI20, so can't port it myself and haven't heard from anyone to do it instead.

soocki, 2017-01-17:
I will send you one I have one to spare and if I don't get this working it is sort of useless to me anyway. Would you be interested in looking at it for me/us/humanity?

CurlyMoo, 2017-01-18:
Maybe we can contact the original developer and the manufacturer first. The whole idea is to have companies start supported wiringX anyway. That already worked with SolidRun and Odroid who have supported this project massively.

soocki, 2017-01-18:
Curly could you point me to some article or resource that would get me started on low-level gpio access methods you are using in wiringX? Where would be a good place to start, if you don't mind me asking?

CurlyMoo, 2017-01-18:
What i did just now:
1. Search for CI20 in google.
2. Click on second link: MIPS Creator CI20 - eLinux.org
3. The second line says ` It incorporates an Ingenic JZ4780 SoC which includes a 1.2GHz dual core MIPS32 processor and Imagination PowerVR SGX540 GPU`
4. Search for the SoC datasheet on Google: `JZ4780 datasheet`
5. Click on the first link you see which point to the PDF.

Then start reading and puzzling.

closed 2019-11-26 by CurlyMoo

larok00, 2023-03-04:
Hello, I'm too much of a beginner to understand the specifics of adding a new device to the repo (in the way menioned above) but I really don't know of any way to use this device other than wiringX. Is there any hope that this will get added? If not do you know of any other way I could control the GPIO in C or Python? Thanks!