wiringX/wiringX #89: SPI support other than spidev0.{0,1}

Artox, 2017-02-26:
My use-case is Hummingboard 2, spi bus on mikrobus header.
It is spidev1.xyz

I suggest:
int wiringXSPISetup(int device, int channel, int speed);
and then use sprintf to generate the /dev/spidevx.y string

Problem is, how big can the string get?
So new suggestion:
int wiringXSPISetup(unsigned char device, unsigned char channel, int speed);

This way device and channel are maximum 255 (length 3).
So maximum length can be calculated and is not much.

Let me know what you think, I am too busy to make a full patch now.

CurlyMoo, 2019-11-26:
Any chance you can incorporate this in a PR?

closed 2022-10-28 by CurlyMoo

Artox, 2022-10-28:
@CurlyMoo Hey, has this one really been completed?

CurlyMoo, 2022-10-28:
So, but the last request hasn't been addressed for 3 years.

Artox, 2022-10-28:
Indeed. I completely forgot about it, and it doesn't look like someone else was asking.
I was cautious about the change, since it changes API - and I didn't feel so great about breaking any existing users.

CurlyMoo, 2022-10-28:
A PR improves the discussion. And API changes can be dealt with with well placed defines.