wiringX/wiringX #93: problem with gpio.so on hummingboard

HammerPiano, 2017-03-10:
I found out about your library, and it has problem making it a package. I am running it on a Hummingboard i2ex with debian GNU/linux 7.11 (wheezy)
When i try to make it into a package, it can't find gpio.so file, I didnt get it when i downloaded the library.
Where can i find it?

CurlyMoo, 2017-03-12:
Can you post full shell output.

HammerPiano, 2017-03-12:
Actually i managed to solve my problem, i had to make the .so file from the wiringX.c and the hummingboard.c files. After I packed everything, I managed to blink a led :D
And an other question(or should I open it in another issue?) : I want to do i2c communications with my arduino, how to I do that with your library?

CurlyMoo, 2017-03-12:
We're offering a low-level IO library. How i2c works can be found elsewhere, i can't help you with it.

closed 2019-11-26 by CurlyMoo