wiringX/wiringX #97: Radxa Rock Python Module Installation

8BitMicro, 2017-04-20:
Hello all, as the title thread suggests, I am trying to figure out the installation process for the WiringX Python module. So far, all of my dealings with WiringX have been in applications where I was coding in C. 

My background is in embedded applications with 8-bit microcontrollers and DSPs, so I have limited embedded Linux knowledge. I am using a Radxa Rock and running Lubuntu. If anybody out there could help me figure out the procedure for installing this module, I would be most appreciative. Thanks so much for your time everyone and have a great day.

pilino1234, 2017-04-20:
Check the readme.txt in the `python/` directory, it says what you need to do:

https://github.com/wiringX/wiringX/blob/master/python/README.txt

8BitMicro, 2017-04-20:
Thanks for your time Pilino1234, I have tried the installation process you recommended and have been stuck at calling "ADD_LIBRARY for library sources without any sources files." 

Cmake Error: Cmake can not determine linker language for target: wiringx_shared
Cmake Error: Cmake can not determine linker language for target: wiringx_shared
Cmake Error: Cannot determine link language for target "wiringx_shared"
Cmake Error: Cannot determine link language for target "wiringx_static"
Cmake Error: Cmake can not determine linker language for target: wiringx_static

pilino1234, 2017-04-20:
Which commands did you run?

8BitMicro, 2017-04-20:
Hey Pilino, I appreciate the help! So far, I've just run cd build and "cmake .. " (after place what I thought were all the relevant C files into the build file - they are also located in my home directory). Thanks for your help.

pilino1234, 2017-04-20:
I'll need some more details. Can you start with a fresh clone, and follow the instructions step by step? Note that you have to start the installation of the python-wiringx module **in the `python/` dir.**

8BitMicro, 2017-04-20:
Thanks so much for your help pilino! The module has no been succesfully installed onto my board, but I have one last question if that's okay with you. It appears that one of the attributes I am calling (wiringXISR and waitForInterrupt) no longer exist (I just read the list of supported functions, but the interrupt.py file in the examples still lists them). Are these no longer supported? Thanks.

pilino1234, 2017-04-20:
Nice to hear that you got it working! Feel free to close this issue 😃

closed 2022-10-28 by CurlyMoo