wiringX/wiringX #98: Just 8 valid pins in HummingBoard Edge

David-Proconsi, 2017-05-09:
Hi,

I have a Hummingboard Edge i4Pro i.MX6 Quad

When using WiringX in Python for HummingBoard Edge it lets me configure pins from 0 to 7 while it is supposed to have 33.

```python
from wiringX import gpio
gpio.setup()
pines = [gpio.valid(n) for n in range(200)]
pines.count(True)
```
>  8

That eight work but the mapping is not as in de documentation.

Is it an Issue or i am forgetting something? I have used wiringX in a Raspberry Pi succesfuly.

CurlyMoo, 2017-05-09:
Are you using the rewrite branch?

David-Proconsi, 2017-05-10:
Thanks for the soon replay.
Sorry, I was using master :(

But it does not work with python right now. 
I solved it and opened the issue #99 for that.

closed 2017-05-10 by David-Proconsi