wiringX/wiringX #99: Python ImportError in Hummingboard

David-Proconsi, 2017-05-10:
```python
from wiringX import gpio
```
>ImportError: /usr/local/lib/python2.7/dist-packages/wiringX/gpio.so: undefined symbol: raspberrypizeroInit

I solved it adding in python/setup.py

```
'../src/platform/raspberrypi/raspberrypizero.c',
```

Like that:

```
'../src/platform/raspberrypi/raspberrypi2.c',
'../src/platform/raspberrypi/raspberrypi3.c',
'../src/platform/raspberrypi/raspberrypizero.c',
'../src/platform/solidrun/hummingboard_base_pro_dq.c',
'../src/platform/solidrun/hummingboard_base_pro_sdl.c',
```

Then compile and reinstall.

I think there are more missing references to raspberrypizero

CurlyMoo, 2017-05-10:
Can you do a PR with your findings?

David-Proconsi, 2017-05-11:
I tried but...

[image]

pilino1234, 2017-05-11:
You need to fork the pilight repo to your own Github account first, push the changes there, and then make a PR. See https://guides.github.com/activities/forking/

David-Proconsi, 2017-05-11:
Thanks, I use git everyday but this is the first time I collaborating in a public project.

closed 2017-05-12 by David-Proconsi