wiringX/wiringX #103: Logo Design for WiringX

DRSDavidSoft, 2017-12-01:
Hi again

### I decided to design a logo for the `wiringX` project, considering that the GitHub page had none.

[image]

### And a minimal design:

[image]

<hr>

### What do you think? Is it good enough to be used here?

[image]

pilino1234, 2017-12-01:
I like it, 👍 from me!

CurlyMoo, 2017-12-02:
Great that you are willing to put your time into making a logo! But to be honest, it reminds me of:
`¯\_(ツ)_/¯`

I don't think that covers wiringX :smile:

closed 2022-10-28 by CurlyMoo