wiringX/wiringX #105: Raspberry pi 3b+ support

louis-lau, 2018-04-04:
I just switched from a 1b+ to a 3b+. Pilight has stopped working and there's absolutely no errors. Using gpio-platform raspberrypi3.

Is the 3b+ not supported yet? or am I going crazy?

Thanks

CurlyMoo, 2018-04-04:
Is that the latest Raspberry Pi?

louis-lau, 2018-04-04:
Yup, released a couple of weeks ago :)
https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/

Pinout is exactly the same as the 3b. It also uses the same processor at a higher clockspeed.

CurlyMoo, 2018-04-04:
Can you use the platform 'raspberrypi3' and check if that works?

louis-lau, 2018-04-04:
Did that, it didn't.

CurlyMoo, 2018-04-04:
Can you eloborate?

louis-lau, 2018-04-04:
Sure, this is the pilight config I'm using:
```
{
        "devices": {},
        "rules": {},
        "gui": {},
        "settings": {
                "log-level": 6,
                "pid-file": "/var/run/pilight.pid",
                "log-file": "/var/log/pilight.log",
                "webserver-enable": 1,
                "webserver-root": "/usr/local/share/pilight/webgui",
                "webserver-http-port": 5001,
                "webserver-cache": 1,
                "gpio-platform": "raspberrypi3",
                "port": 5000
        },
        "hardware": {
                "433gpio": {
                        "sender": 0,
                        "receiver": 1
                }
        },
        "registry": {
                "webserver": {
                        "ssl": {
                                "certificate": {
                                        "secure": 0,
                                        "location": "/etc/pilight/pilight.pem"
                                }
                        }
                },
                "pilight": {
                        "version": {
                                "current": "8.0.10"
                        }
                }
        }
```

I have a RF thing connected to wiringpi gpio 0. When using the send command it looks successful (no errors), but nothing happens on my switches.

Then changed back the platform to raspberrypi1b+ and put it back into my old pi just to be sure. Using the exact same gpio pins it works fine.

louis-lau, 2018-04-04:
Nothing in the logs btw, any debugging I can do?

CurlyMoo, 2018-04-04:
Can you check using the wiringX examples?

CurlyMoo, 2018-04-04:
Ofc, you need a LED or something to actually see if the GPIO works.

louis-lau, 2018-04-04:
Alright fuck. I ripped a led out of an old appliance, and it's working...

Any clues as to why the RF module isn't working with pilight?

louis-lau, 2018-04-04:
(it's one of these)
[image]

louis-lau, 2018-04-04:
Well shit, looks like it was my power supply.
Thanks a lot for your time!

At least now you know for sure it's compatible with the 3b+ 😄

closed 2018-04-04 by louis-lau