wiringX/wiringX #110: python import error on hummingboard: gpio.so: undefined symbol: raspberrypi4Init

spotrevoc, 2019-10-21:
The c library works just fine with the build but when I try to use python (after install of course), I get the following error when I attempt to do "from wiringX import gpio":

Traceback (most recent call last):
  File "new.py", line 1, in <module>
    from wiringX import gpio
ImportError: /usr/local/lib/python2.7/dist-packages/wiringX/gpio.so: undefined symbol: raspberrypi4Init

I first built/compiled as is from the git repo. After failure, I attempted to go through and remove all reference to the raspberry pi 4 and ended up with the same result.  I then tried to go through and reference the src (.c and .h) files to the raspberry pi 4 anywhere references appeared to be missing from the original code, same result. 

Has anyone had success with the python version on hummingboard? What am I missing here?

CurlyMoo, 2019-10-22:
It should work after the raspberry pi 4 has been added to the python code.

spotrevoc, 2019-10-22:
Is that going to be an upstream patch to the python? Is there a fix i can
do to get it working without the rpi4 references?

CurlyMoo, 2019-10-22:
Just download the latest source and try again just as you did earlier.

closed 2019-11-26 by CurlyMoo