wiringX/wiringX #112: python3 support

aadityabagga, 2019-11-29:
Hi!

With only python3 installed, when I tried to build the python bindings, there is no output:
~~~~
aaditya ~/git/wiringX/python/build $ cmake ..
-- The C compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aaditya/git/wiringX/python/build
aaditya ~/git/wiringX/python/build $ make
aaditya ~/git/wiringX/python/build $ cpack -G RPM
CPack: Create package using RPM
CPack: Install projects
CPack: - Run preinstall target for: wiringX
CPack: - Install project: wiringX []
CMake Error at /home/aaditya/git/wiringX/python/build/cmake_install.cmake:45 (file):
  file INSTALL cannot find "/wiringX/gpio.so": No such file or directory.

CMake Error at /home/aaditya/git/wiringX/python/build/cmake_install.cmake:49 (file):
  file INSTALL cannot find "/wiringX/__init__.py": No such file or directory.

CPack Error: Error when generating package: python-wiringX
~~~~

after installing Python2 it work:
~~~~
aaditya ~/git/wiringX/python/build $ cmake ..
In file included from /usr/include/python2.7/Python.h:8,
                 from wiringX/wiringx.c:25:
/usr/include/python2.7/pyconfig.h:1223: warning: "_POSIX_C_SOURCE" redefined
 1223 | #define _POSIX_C_SOURCE 200112L
      | 
In file included from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from wiringX/wiringx.c:9:
/usr/include/features.h:266: note: this is the location of the previous definition
  266 | # define _POSIX_C_SOURCE 200809L
      | 
wiringX/wiringx.c: In function ‘py_setup’:
wiringX/wiringx.c:308:28: warning: passing argument 2 of ‘wiringXSetup’ from incompatible pointer type [-Wincompatible-pointer-types]
  308 |  if(wiringXSetup(platform, py_wiringXLog) < 0){
      |                            ^~~~~~~~~~~~~
      |                            |
      |                            void (*)(int,  const char *, ...)
In file included from wiringX/wiringx.c:27:
../src/wiringx.h:71:37: note: expected ‘void (*)(int,  char *, int,  const char *, ...)’ but argument is of type ‘void (*)(int,  const char *, ...)’
   71 | int wiringXSetup(char *name, void (*func)(int, char *, int, const char *, ...));
      |                              ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/wiringx.c: In function ‘wiringXSupportedPlatforms’:
../src/wiringx.c:731:13: warning: unused variable ‘x’ [-Wunused-variable]
  731 |  int i = 0, x = 0;
      |             ^
In file included from ../src/wiringx.c:24:
../src/wiringx.c: At top level:
../src/i2c-dev.h:36:14: warning: inline function ‘i2c_smbus_write_word_data’ declared but never defined
   36 | inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:35:14: warning: inline function ‘i2c_smbus_read_word_data’ declared but never defined
   35 | inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:34:14: warning: inline function ‘i2c_smbus_write_byte_data’ declared but never defined
   34 | inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:33:14: warning: inline function ‘i2c_smbus_read_byte_data’ declared but never defined
   33 | inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:32:14: warning: inline function ‘i2c_smbus_write_byte’ declared but never defined
   32 | inline __s32 i2c_smbus_write_byte(int fd, int value);
      |              ^~~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:31:14: warning: inline function ‘i2c_smbus_read_byte’ declared but never defined
   31 | inline __s32 i2c_smbus_read_byte(int fd);
      |              ^~~~~~~~~~~~~~~~~~~
../src/i2c-dev.h:30:14: warning: inline function ‘i2c_smbus_access’ declared but never defined
   30 | inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
      |              ^~~~~~~~~~~~~~~~
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aaditya/git/wiringX/python/build
aaditya ~/git/wiringX/python/build $ make
aaditya ~/git/wiringX/python/build $ cpack -G RPM
CPack: Create package using RPM
CPack: Install projects
CPack: - Run preinstall target for: wiringX
CPack: - Install project: wiringX []
CPack: Create package
CPackRPM: Will use GENERATED spec file: /home/aaditya/git/wiringX/python/build/_CPack_Packages/Linux/RPM/SPECS/python-wiringx.spec
CPack: - package: /home/aaditya/git/wiringX/python/build/python-wiringx-245-g8efb389.rpm generated.

~~~~
Does WiringX support Python3? I am a new comer to wiringX and Python, request pardon for newbie mistakes.

Tested this on a recently setup slarm64 (unofficial Slackware ARM 64).
Thanks!

CurlyMoo, 2019-11-29:
I don't know. I'm not using the python bindings myself.

aadityabagga, 2019-11-29:
> I don't know. I'm not using the python bindings myself.

Okk, will keep this issue open and check if it works with Python3.

aadityabagga, 2019-11-29:
I think we will need to update this file:
https://github.com/wiringX/wiringX/blob/master/python/CMakeLists.txt

Currently it seems to search and install only for Python2.

CurlyMoo, 2019-11-29:
Try it and let me know if it worked.

closed 2022-10-28 by CurlyMoo