wiringX/wiringX #113: wiringx.org domain gone?

felfert, 2020-07-30:
I am trying to get some doc, but resolving the hosts in the links in README.md result in a NXDOMAIN response?!

CurlyMoo, 2020-07-30:
They work fine here?

felfert, 2020-07-30:
Not from Germany. Tested from 3 locations (on each, also a functional request for gmx.net):

1. On listserv.isdn4linux.de (dedicated server, hosted at IONOS (former 1&1)
```
[felfert@listserv ~]$ host -t ns wiringx.org
Host wiringx.org not found: 3(NXDOMAIN)
[felfert@listserv ~]$ host -t ns gmx.net
gmx.net name server ns-gmx.ui-dns.de.
gmx.net name server ns-gmx.ui-dns.com.
gmx.net name server ns-gmx.ui-dns.org.
gmx.net name server ns-gmx.ui-dns.biz.

```
2. On my home PC, connected via Vodafone cable
```
[felfert@fsun ~]$ host -t ns wiringx.org
Host wiringx.org not found: 3(NXDOMAIN)
[felfert@fsun ~]$ host -t ns gmx.net
gmx.net name server ns-gmx.ui-dns.biz.
gmx.net name server ns-gmx.ui-dns.com.
gmx.net name server ns-gmx.ui-dns.de.
gmx.net name server ns-gmx.ui-dns.org.
```
3. At work, connected via german Telekom
```
[felfert@jenkins ~]$ host -t ns wiringx.org
Host wiringx.org not found: 3(NXDOMAIN)
[felfert@jenkins ~]$ host -t ns gmx.net
gmx.net name server ns-gmx.ui-dns.com.
gmx.net name server ns-gmx.ui-dns.biz.
gmx.net name server ns-gmx.ui-dns.de.
gmx.net name server ns-gmx.ui-dns.org.

```
Reported down from US as well:
https://www.websiteplanet.com/webtools/down-or-not/?dorn=wiringx.org

CurlyMoo, 2020-07-30:
Indeed: https://www.whatsmydns.net/#A/wiringx.org

felfert, 2020-07-31:
Again: https://www.whatsmydns.net/#A/wiringx.org

closed 2022-10-28 by CurlyMoo