wiringX/wiringX #117: gpio.so missing

flaep, 2022-12-10:
Hi,

I tried to compile the python module but get the following error when creating the deb file

> 
> rock-3a:build:# cpack -G DEB                                                                                                   <master>
> CPack: Create package using DEB
> CPack: Install projects
> CPack: - Run preinstall target for: wiringX
> CPack: - Install project: wiringX []
> CMake Error at /root/wiring/wiringX/python/build/cmake_install.cmake:50 (file):
>   file INSTALL cannot find "/wiringX/gpio.so": No such file or directory.

CurlyMoo, 2022-12-10:
I've got no issues what so ever:
```
root@dev:/root/wiringX/build# cmake ..
-- The C compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/wiringX/build
root@dev:/root/wiringX/build# make
Scanning dependencies of target wiringx_shared
[  1%] Building C object CMakeFiles/wiringx_shared.dir/src/i2c-dev.c.o
[  2%] Building C object CMakeFiles/wiringx_shared.dir/src/wiringx.c.o
In file included from /root/wiringX/src/wiringx.c:24:0:
/root/wiringX/src/i2c-dev.h:36:14: warning: inline function ‘i2c_smbus_write_word_data’ declared but never defined
 inline __s32 i2c_smbus_write_word_data(int fd, int cmd, __u16 value);
              ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:35:14: warning: inline function ‘i2c_smbus_read_word_data’ declared but never defined
 inline __s32 i2c_smbus_read_word_data(int fd, int cmd);
              ^~~~~~~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:34:14: warning: inline function ‘i2c_smbus_write_byte_data’ declared but never defined
 inline __s32 i2c_smbus_write_byte_data(int fd, int cmd, int value);
              ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:33:14: warning: inline function ‘i2c_smbus_read_byte_data’ declared but never defined
 inline __s32 i2c_smbus_read_byte_data(int fd, int cmd);
              ^~~~~~~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:32:14: warning: inline function ‘i2c_smbus_write_byte’ declared but never defined
 inline __s32 i2c_smbus_write_byte(int fd, int value);
              ^~~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:31:14: warning: inline function ‘i2c_smbus_read_byte’ declared but never defined
 inline __s32 i2c_smbus_read_byte(int fd);
              ^~~~~~~~~~~~~~~~~~~
/root/wiringX/src/i2c-dev.h:30:14: warning: inline function ‘i2c_smbus_access’ declared but never defined
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^~~~~~~~~~~~~~~~
[  3%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/platform.c.o
[  4%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/linksprite/pcduino1.c.o
[  6%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/lemaker/bananapi1.c.o
[  7%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/lemaker/bananapim2.c.o
[  8%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/solidrun/hummingboard_base_pro_dq.c.o
[  9%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/solidrun/hummingboard_base_pro_sdl.c.o
[ 10%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/solidrun/hummingboard_gate_edge_dq.c.o
[ 12%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/solidrun/hummingboard_gate_edge_sdl.c.o
[ 13%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi1b+.c.o
[ 14%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi1b1.c.o
[ 15%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi1b2.c.o
[ 17%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi2.c.o
[ 18%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi3.c.o
[ 19%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypi4.c.o
[ 20%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/raspberrypi/raspberrypizero.c.o
[ 21%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/hardkernel/odroidc1.c.o
[ 23%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/hardkernel/odroidc2.c.o
[ 24%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/hardkernel/odroidxu4.c.o
[ 25%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/xunlong/orangepipc+.c.o
[ 26%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/xunlong/orangepipc2.c.o
[ 28%] Building C object CMakeFiles/wiringx_shared.dir/src/platform/radxa/rock4.c.o
[ 29%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/soc.c.o
[ 30%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/allwinner/a10.c.o
[ 31%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/allwinner/a31s.c.o
[ 32%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/allwinner/h3.c.o
[ 34%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/allwinner/h5.c.o
[ 35%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/nxp/imx6dqrm.c.o
[ 36%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/nxp/imx6sdlrm.c.o
[ 37%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/broadcom/2711.c.o
[ 39%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/broadcom/2835.c.o
[ 40%] Building C object CMakeFiles/wiringx_shared.dir/src/soc/broadcom/2836.c.o
[... 107 lines cut ...]
/root/wiringX/src/soc/rockchip/../../i2c-dev.h:30:14: warning: inline function ‘i2c_smbus_access’ declared but never defined
 inline __s32 i2c_smbus_access(int fd, char rw, int cmd, int size, union i2c_smbus_data *data);
              ^~~~~~~~~~~~~~~~
[ 92%] Linking C static library libwiringx.a
[ 92%] Built target wiringx_static
Scanning dependencies of target wiringx-blink
[... cut ...]
[ 97%] Built target wiringx-interrupt
Scanning dependencies of target wiringx-read
[ 98%] Building C object CMakeFiles/wiringx-read.dir/examples/read.c.o
[100%] Linking C executable wiringx-read
[100%] Built target wiringx-read
root@dev:/root/wiringX/build# cpack -G DEB
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: wiringx
CPack: - Install project: wiringx
CPack: Create package
CPack: - package: /root/wiringX/build/libwiringx_250-gc9ade41.deb generated.
```

flaep, 2022-12-11:
Hi @CurlyMoo Iam trying to complile python-wiringx*.deb according to the readme in wiringX/python. Not the main wiringX.

flaep, 2022-12-11:
looks like wiringx has no code for the radxa rock3a. :(

CurlyMoo, 2022-12-11:
You can ask @RadxaYuntian about it.

flaep, 2022-12-11:
since it compiled now, just not as described in the readme the issue can be closed. 
I asked in the radxa forum https://forum.radxa.com/t/wiringx-on-rock3a/13419 and might create a new issue

CurlyMoo, 2022-12-11:
Did you miss there is a seperate README in the python folder?
https://github.com/wiringX/wiringX/tree/master/python

CurlyMoo, 2022-12-11:
Don't make th build folder inside the python folder, but inside the root of wiringX.

flaep, 2022-12-11:
but does also build the python package?

CurlyMoo, 2022-12-11:
Hmm, good question 😆