wiringX/wiringX #119: PMUCRU / CRU address space

mattjlewis, 2023-01-09:
Should this be writing to the PMUCRU address space (0xff75000) rather than CRU (0xff760000) for GPIO banks 0 & 1?
https://github.com/wiringX/wiringX/blob/master/src/soc/rockchip/rk3399.c#L390

CurlyMoo, 2023-01-10:
@RadxaYuntian @nascs any comments?

nascs, 2023-01-11:
In fact, he's right.

mattjlewis, 2023-01-11:
FYI spotted while adding [RK3299 support](https://github.com/mattjlewis/diozero/blob/main/diozero-core/src/main/java/com/diozero/internal/board/soc/rockchip/Rockchip3399MmapGpio.java) in my [diozero](https://www.diozero.com) library. I only came across your excellent implementation midway through adding support for Rock 4C+.

Out of interest, do you plan on adding PU/PD support? I have [attempted to implement](https://github.com/mattjlewis/diozero/blob/main/diozero-core/src/main/java/com/diozero/internal/board/soc/rockchip/Rockchip3399MmapGpio.java#L367) as per the data sheet and can see it works only for some pins (> 148 from memory).

mattjlewis, 2023-01-11:
diozero intends to expose a higher-level device abstraction. I need to consider whether I can use wiringX under the hood to avoid having to implement mmap GPIO support myself - which is quite time consuming not to mention complex and expensive to test!