wiringX/wiringX #122: merge wiringX into buildroot mainline

gtxzsxxk, 2023-09-21:
I am presently working on Buildroot and have come across your remarkable work, which appears to be highly compatible and helpful for the embedded rootfs distributions. I am considering merging your work into the mainline Buildroot.

My plan is to include wiringX as one of the Target Packages in buildroot, and the corresponding options would be displayed when supported boards or SoCs are selected. From my understanding, there shouldn't be any licensing conflicts, so would I be permitted to proceed with this? I eagerly await your response.

CurlyMoo, 2023-09-21:
Without knowing the license of the target project i can't say anything about it.

gtxzsxxk, 2023-09-21:
If you are meaning that the target is the buildroot, the  license is GPLv2.

CurlyMoo, 2023-09-21:
If the target project is GPLv2 than it's fine. As long as you make all changes you make to wiringX open source.

gtxzsxxk, 2023-09-21:
Sure, I will follow that. Thanks for explanation.

closed 2023-09-21 by gtxzsxxk