wiringX/wiringX #123: Build failure on rockpi-4-c-plus Debian 11.8

m1larepa, 2023-10-18:
radxa@rock-4c-plus:~/src/wiringX/build$ LANG=C make
[ 37%] Built target sources
[ 43%] Linking C shared library libwiringX.so
/usr/bin/ld: CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x78): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x70): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x78): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x70): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x270): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x268): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xc0): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xb8): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x30): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x38): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x8): multiple definition of `ci20'; CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x68): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x10): multiple definition of `radxa'; CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x260): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x18): multiple definition of `bananapi'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x470): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x20): multiple definition of `raspberrypi'; CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xb0): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x28): multiple definition of `hummingboard'; CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x68): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/wiringx_shared.dir/build.make:99: libwiringX.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:216: CMakeFiles/wiringx_shared.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
radxa@rock-4c-plus:~/src/wiringX/build$

m1larepa, 2023-10-18:
Well, it's now OK after remove and reinstall from git ...
Sorry

closed 2023-10-18 by m1larepa

kalyanikasam, 2024-06-03:
Hi, i am trying to install wiringX on radxa's rock5b following the instructions in readme and i am getting same errors as above, i tried reinstalling but same errors.
gcc version 11.4.0 ()ubuntu 22.04), cmake version 3.22.1, GNU Make 4,3.
please help me fix multiple definition error

CurlyMoo, 2024-06-03:
Have you tried the same solution as @m1larepa ?

kalyanikasam, 2024-06-04:
Yes i have removed, cloned again and build but same error comes

/usr/bin/ld: CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x78): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x70): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x78): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x70): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x270): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x268): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xc0): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xb8): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x30): multiple definition of `platforms'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x478): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x38): multiple definition of `wiringXLog'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x480): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x8): multiple definition of `ci20'; CMakeFiles/sources.dir/src/ci20.c.o:(.bss+0x68): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x10): multiple definition of `radxa'; CMakeFiles/sources.dir/src/radxa.c.o:(.bss+0x260): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x18): multiple definition of `bananapi'; CMakeFiles/sources.dir/src/bananapi.c.o:(.bss+0x470): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x20): multiple definition of `raspberrypi'; CMakeFiles/sources.dir/src/raspberrypi.c.o:(.bss+0xb0): first defined here
/usr/bin/ld: CMakeFiles/sources.dir/src/wiringX.c.o:(.bss+0x28): multiple definition of `hummingboard'; CMakeFiles/sources.dir/src/hummingboard.c.o:(.bss+0x68): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/wiringx_shared.dir/build.make:92: libwiringX.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/wiringx_shared.dir/all] Error 2
make: *** [Makefile:156: all] Error 2