wiringX/wiringX #126: WiringX incompatible with Linux raspberrypi 6.6.20+rpt-rpi-v

clentfort, 2024-03-31:
Hi,

I'm running into issues with WiringX (used through Pilight, thanks for both projects btw).

I'm seeing the following erros

> [Mar 31 09:10:50:985819] pilight-daemon: ERROR: The Broadcom 2711 failed to write to /sys/class/gpio/export for gpio exporting (Invalid argument)
> [Mar 31 09:10:50:985861] pilight-daemon: ERROR: unable to register interrupt for pin 2

From my research the `/sys/class/gpio/export` has been deprecated and was removed in kernel 6.620. 

I'd happy to help update this library to use [newer APIs](https://docs.kernel.org/userspace-api/gpio/chardev.html) but I'm not sure how to start this migration. Any advice would be appreciated.

CurlyMoo, 2024-03-31:
That would be great. Preferably with a kernel version check so it picks the right approach based on the kernel version.

Easiest would just be to search the source for the `/sys/class/gpio` path and update those location using it.

clentfort, 2024-03-31:
Hi thanks for the fast reply.

Unfortunately I have zero experience in systems/c-programming and would require some more hand-holding.

From glancing at the code and the new API I would first need to identify which gpio chip to use, select it and the use `ioctl` to reconfigure the lines?

For anyone (potentially me) running into this problem again in the future:

For now to get pilight up and running again I downgraded the kernel back to `Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux` by running `apt remove linux-image-6.6.20+rpt-rpi-2712 linux-image-6.6.20+rpt-rpi-v8`, accepting the consequences of removing the running kernel followed by `apt-get reinstall linux-image-6.1.0-rpi{7,8}-rpi-2712 linux-image-6.1.0-rpi{7,8}-rpi-v8` and only then a `reboot`.

CurlyMoo, 2026-02-08:
@philn56 Would indeed be great if you could make that PR.

CurlyMoo, 2026-02-08:
This doesn't identify which kernel the RPi is running. Can you include that? So that in case of the older kernel, it uses the older code and based on the newer kernel the new code.

CurlyMoo, 2026-02-09:
So this receives the indexes from the kernel itself? Does that indeed work on older kernels?

CurlyMoo, 2026-02-09:
Now others can read this and volunteer to test 😄