wiringX/wiringX #130: undefined symbol: rockchipSelectableFd

anotherche, 2025-09-02:
I am trying to use python wiringX for Rockpi 4a. After compiling the source code according to two instructions (for wiringx and for python part), the python examples do not run due to the error 
```
undefined symbol: rockchipSelectableFd
``` 
which occurs when importing `from wiringX import gpio`. This function is indeed not defined in the gpio.so library (nm -D gpio.so returns U for rockchipSelectableFd)
```
000000000006e3f8 B rock4
000000000002db64 T rock4Init
000000000006e400 B rock5b
000000000002dc74 T rock5bInit
                 U rockchipDigitalRead
                 U rockchipGC
                 U rockchipGetLayout
                 U rockchipGetPinName
                 U rockchipISR
                 U rockchipSelectableFd
                 U rockchipSetIRQ
                 U rockchipSetMap
                 U rockchipWaitForInterrupt

```
Perhaps some dynamic import from another library is broken? Which one? 

At the same time, the *.c examples (like wiringx-blink or wiringx-read) run and work normally, and it is possible to read and write to different pins.

CurlyMoo, 2025-09-02:
Can you do a PR for these missing sources?

CurlyMoo, 2025-09-02:
No hurry. I always want to encourage those who found bugs and the fixes to do a PR themselves.